ukama/ukama #479: Implement a service  for document and image upload to Amazon S3

Brackleycassinga, 2023-03-30:
## Description:

We need to implement a gRPC service in Go that will allow users to upload documents and images to Amazon S3 and return a URL with the uploaded file. This service will be used to provide file upload functionality to our application.

The service should accept the following inputs:

- [ ] File data: The binary data of the file to be uploaded.
- [ ] File name: The name of the file to be uploaded.
- [ ] Bucket name: The name of the S3 bucket where the file should be uploaded.
- [ ] Access key ID: The access key ID for the AWS account.
- [ ] Secret access key: The secret access key for the AWS account.

delvatt, 2023-04-03:
I think this issue should  not be limited only for user uploads, but also for files that might be generated for the users, e.g invoices and also files that might be sent to us from some of our providers e.g CDR files. #470

markhor, 2023-04-03:
> I think this issue should not be limited only for user uploads, but also for files that might be generated for the users, e.g invoices and also files that might be sent to us from some of our providers e.g CDR files. #470

And probably something to go with the "common services"

closed 2025-01-08 by Brackleycassinga